luigifreda/pyslam #150: FAST ORB feature tracker

ufukasia, 2025-03-01:
# FeatureTrackerConfigs: SHI_TOMASI_ORB, FAST_ORB, ORB, ORB2, ORB2_FREAK, ORB2_BEBLID, BRISK, AKAZE, FAST_FREAK, SIFT, ROOT_SIFT, SURF, KEYNET, SUPERPOINT, CONTEXTDESC, LIGHTGLUE, XFEAT, XFEAT_XFEAT
    # WARNING: At present, SLAM does not support LOFTR and other "pure" image matchers (further details in the commenting notes about LOFTR in feature_tracker_configs.py).
    feature_tracker_config = FeatureTrackerConfigs.ORB2

"In this snippet, Fast ORB is mentioned as the feature tracker, but Fast ORB isn't working. Is there any chance it might work?"

luigifreda, 2025-03-01:
Hi,
I can use `FAST_ORB` without any issues. 
Could you share your error message and all the relevant messages related to that?

closed 2025-03-01 by luigifreda

ufukasia, 2025-03-01:
Tomorrow when I get home, I'll try again. It might be my mistake. Thanks a lot for everything.

ufukasia, 2025-03-02:
not working with fast orb, i installed newly 

(pyslam) (base) masa24@masa24-MS-7C56:~/Desktop/pyslam$ /home/masa24/anaconda3/envs/pyslam/bin/python /home/masa24/Desktop/pyslam/main_slam.py
 Using stereo settings file: /home/masa24/Desktop/pyslam/settings/EuRoC_stereo.yaml
------------------------------------
2025-03-02 17:31:11.799791: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:477] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
E0000 00:00:1740925871.816558    8216 cuda_dnn.cc:8310] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
E0000 00:00:1740925871.821878    8216 cuda_blas.cc:1418] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2025-03-02 17:31:11.839381: I tensorflow/core/platform/cpu_feature_guard.cc:210] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
 WARNING: cannot import VolumetricIntegratorGaussianSplatting from volumetric_integrator_gaussian_splatting, check the file docs/TROUBLESHOOTING.md
 Using stereo settings file: /home/masa24/Desktop/pyslam/settings/EuRoC_stereo.yaml
------------------------------------
 dataset_factory - sensor_type: STEREO
 dataset_factory - start_frame_id: 0
 [EurocDataset] automatically rectifying the stereo images
Processing Euroc Sequence of lenght:  2032
Trajectory file results/metrics_20250302_173115/trajectory_online.txt opened
Trajectory file results/metrics_20250302_173115/trajectory_final.txt opened
 groundtruth_factory - start_frame_id: 0
using groundtruth:  euroc
base_path:  /media/masa24/Seagate Backup/sdata/MH04/mav0/state_groundtruth_estimate0
Processing Euroc groundtruth of lenght:  19753
[associate] Number of matches: 2033
 WARNING: Using stereo camera, images are automatically rectified, and DistCoef is set to [0,0,0,0,0]
 feature_tracker_config:  {'num_features': 1200, 'num_levels': 8, 'scale_factor': 1.2, 'detector_type': FeatureDetectorTypes.FAST, 'descriptor_type': FeatureDescriptorTypes.ORB, 'sigma_level0': 1.0, 'match_ratio_test': 0.7, 'tracker_type': FeatureTrackerTypes.DES_BF}
 loop_detection_config:  {'global_descriptor_type': GlobalDescriptorType.DBOW3, 'local_feature_manager_config': None, 'local_descriptor_aggregation_type': LocalDescriptorAggregationType.DBOW3, 'vocabulary_data': <loop_detector_vocabulary.DBow3OrbVocabularyData object at 0x7b6600951c30>}
FeatureManager: num_levels:  8
FeatureManager: Using opencv  4.10.0
 WARNING: using NON-ORIENTED features:  FAST - ORB  (i.e. kp.angle=0)
FeatureManager: num_levels:  8
 matcher: BfFeatureMatcher - norm_type: 6, cross_check: False, ratio_test: 0.7
DBoW3::Vocabulary loaded from /home/masa24/Desktop/pyslam/loop_closing/../data/ORBvoc.dbow3 (971814 entries)
Viewer3D: Groundtruth shape: (2032, 3)
Mplot2d: starting the process on figure: 1
Mplot2d: backend Qt5Agg
Mplot2d: starting the process on figure: 2
Mplot2d: backend Qt5Agg
Mplot2d: starting the process on figure: 3
Mplot2d: backend Qt5Agg
Mplot2d: starting the process on figure: 4
Mplot2d: backend Qt5Agg
..................................
getImageRight:  1403638127295097088.png
image: 0, timestamp: 1403638127.295097, duration: 0.04999995231628418
 @tracking STEREO, img id: 0, frame id: 0, state: NO_IMAGES_YET
img.shape: (480, 752, 3), camera: 480x752
img_right.shape:  (480, 752, 3)
timestamp:  1403638127.295097
 WARNING: could not set nFeatures in detector:  FAST
PyramidAdaptor #levels: 8 (from 0 ), scale_factor: 1.2 , sigma0: 1.0 , type: RESIZE
PyramidAdaptor #levels: 8 (from 0 ), scale_factor: 1.2 , sigma0: 1.0 , type: RESIZE
Segmentation fault (core dumped)

luigifreda, 2025-03-02:
I succeeded in reproducing the issue on the same dataset. I got it fixed.