luigifreda/pyslam #151: Issues on Mac

lukasbuehler, 2025-03-06:
Environment: Apple MacBook Pro, M4 Pro, macOS 15.3.1

Installation:`./install_all_venv.sh`. (I also tried `./install_all.sh`, same problems)

# Mutliple modules not found after installation

After running either `./main_slam.py` or `./main_vo.py` I first get multiple warnings:

```
 WARNING: cannot import ORBextractor from orbslam2_features, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import pyslam_utils
 WARNING: cannot import SuperPointFeature2D from feature_superpoint, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import TfeatFeature2D from feature_tfeat, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import Orbslam2Feature2D from feature_orbslam2, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import DelfFeature2D from feature_delf, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import R2d2Feature2D from feature_r2d2, check the file docs/TROUBLESHOOTING.md
 WARNING: cannot import DiskFeature2D from feature_disk, check the file docs/TROUBLESHOOTING.md
```

looking into them, the `thirdparty/orbslam2_features` directory looks ok, but the `lib` subdirectory is empty.

the scripts keep on running for a little while and then I get: ModuleNotFoundError: No module named 'trimesh'

```
Traceback (most recent call last):
  File "/Users/lukas/development/eth/semester_project/experiment-pyslam/./main_vo.py", line 30, in <module>
    from visual_odometry import VisualOdometryEducational
  File "/Users/lukas/development/eth/semester_project/experiment-pyslam/slam/visual_odometry.py", line 24, in <module>
    from feature_tracker import FeatureTrackerTypes, FeatureTrackingResult, FeatureTracker
  File "/Users/lukas/development/eth/semester_project/experiment-pyslam/local_features/feature_tracker.py", line 25, in <module>
    from feature_matcher import feature_matcher_factory, FeatureMatcherTypes
  File "/Users/lukas/development/eth/semester_project/experiment-pyslam/local_features/feature_matcher.py", line 28, in <module>
    from utils_dust3r import Dust3rImagePreprocessor
  File "/Users/lukas/development/eth/semester_project/experiment-pyslam/utilities/utils_dust3r.py", line 23, in <module>
    import trimesh
ModuleNotFoundError: No module named 'trimesh'
```

Maybe this line here is relevant, it is never installed at all, since `CUDA_VERSION` is 0 on Mac:
https://github.com/luigifreda/pyslam/blob/6297dfc97fbfed85298bac7cf36bf88cc3b7e0f3/install_pip3_packages.sh#L159-L169

After that I just installed `trimesh` in the Python virtual environment and ran `./main_slam.py` again, I get the same warnings but this error:

```
Traceback (most recent call last):
  File "/Users/lukas/development/eth/semester_project/pyslam/./main_slam.py", line 32, in <module>
    from slam import Slam, SlamState
  File "/Users/lukas/development/eth/semester_project/pyslam/slam/slam.py", line 41, in <module>
    from loop_closing import LoopClosing
  File "/Users/lukas/development/eth/semester_project/pyslam/loop_closing/loop_closing.py", line 40, in <module>
    from loop_detector_configs import LoopDetectorConfigs
  File "/Users/lukas/development/eth/semester_project/pyslam/loop_closing/loop_detector_configs.py", line 37, in <module>
    from loop_detector_dbow3 import LoopDetectorDBoW3
  File "/Users/lukas/development/eth/semester_project/pyslam/loop_closing/loop_detector_dbow3.py", line 69, in <module>
    class LoopDetectorDBoW3(LoopDetectorBase): 
  File "/Users/lukas/development/eth/semester_project/pyslam/loop_closing/loop_detector_dbow3.py", line 107, in LoopDetectorDBoW3
    def db_query(self, global_des: dbow3.BowVector, frame_id, max_num_results=5): 
AttributeError: module 'pydbow3' has no attribute 'BowVector'
```

Thank you for your help! I will also try to run it on my Linux desktop (with CUDA) tomorrow but this is the state on my Mac. :)

luigifreda, 2025-03-06:
Hi,
I've just verified the install procedure using `install_all.sh` on my macbook (macOS 15.3.1). 
I've resolved the missing `trimesh` package issue. However, everything else works fine on my end. 

Could you please run the following steps?
```bash
./clean.sh --hard   # clean build folders 
./pyenv-delete.sh   # delete the "pyslam" virtual environment
```
Then, open a new terminal and run `install_all.sh` again. 
To assist you effectively, I need the full log of the entire installation process.

lukasbuehler, 2025-03-07:
Thank you for your help, It is working now!
This issue can be clsed now.

Best regards
Lukas

closed 2025-03-07 by lukasbuehler