luigifreda/pyslam #152: lietorch issue

Pravindada, 2025-03-07:
While installing lietorch shows issue related cmake and this error occur while using volumeintegrator gaussian splatting

[image]

luigifreda, 2025-03-07:
Hi,
Thanks for your feedback. Could you provide more details about the issue you're experiencing?
Please take a look at the [guidelines](https://github.com/luigifreda/pyslam/blob/master/docs/TROUBLESHOOTING.md#submitting-a-git-issue) for submitting a Git issue.

Could you share your OS and CUDA versions? This will help in diagnosing the problem more effectively.

Best,
Luigi

luigifreda, 2025-03-07:
Can you try to modify locally this line?
https://github.com/luigifreda/pyslam/blob/master/thirdparty/lietorch/CMakeLists.txt#L33

Replace 
```
if(NVCC_OUTPUT MATCHES "c\\+\\+20" AND GCC_MAJOR_VERSION GREATER_EQUAL 10)
    set(CPP_STANDARD_VERSION "20")
```
with 
```
if(NVCC_OUTPUT MATCHES "c\\+\\+20" AND GCC_MAJOR_VERSION GREATER_EQUAL 10)
    set(CPP_STANDARD_VERSION "17")
```

Then open a terminal, get into the root folder of your pySLAM working copy and run:
```
. pyenv-activate.sh 
cd thirdparty/lietorch
rm -Rf build 
./build.sh
./run-tests.sh   # if everything builds correctly 
```

luigifreda, 2025-03-08:
Hi,
I was able to reproduce the issue on my end. To resolve it, I ran the following command:
```bash
sudo apt-get install -y nvidia-cuda-dev
```

To fix this on your side, please follow these steps:
- Keep the modification I shared in my previous message.
- Open a new terminal and navigate to the root folder of pyslam.
- Run the following commands:
```bash
sudo apt-get install -y nvidia-cuda-dev
.  pyenv-activate.sh     # We are in the root folder of your pySLAM working copy
cd thirdparty/lietorch
rm -Rf build
./build.sh
```

Let me know if this works for you!

closed 2025-03-08 by luigifreda