luigifreda/pyslam #156: for replica

zyh778, 2025-03-26:
The replica dataset I downloaded was lost due to my mistake ，When I download it again for this link:wget https://cvg-data.inf.ethz.ch/nice-slam/data/Replica.zip （only for 11.59GB）
There is only the result part in the dataset, and the RGB image part of colors is lost
I don't know if you have any copies of the colors, but I would appreciate it if you could provide it to me

luigifreda, 2025-03-26:
Hi,
Did you try the download script available here?
https://github.com/facebookresearch/Replica-Dataset

zyh778, 2025-03-26:
> Hi, Did you try the download script available here? https://github.com/facebookresearch/Replica-Dataset

Hi, Yes, I've tried this download script already. However, if I directly use the original dataset or convert it myself to formats like RGB, is it likely to cause unnecessary errors? Could you please share the dataset you used, perhaps via a Google Drive link or another method? I would be extremely grateful.

luigifreda, 2025-03-26:
Hi,
The RGB images come in the same folder as the depth images.
For instance, I have 
```
├── office
         ├── results
                  ├── depth000000.png
                   ....
                  ├── depth001999.png
                  ├── frame000000.jpg
                   ....
                  ├── frame001999.jpg
...
```
To be sure, could you double-check that?

I've just tried to download with 
```
wget https://cvg-data.inf.ethz.ch/nice-slam/data/Replica.zip
```
Then I extracted the content from the zip file, and I found everything that is needed therein.

closed 2025-03-26 by luigifreda

zyh778, 2025-03-26:
> Hi, The RGB images come in the same folder as the depth images. For instance, I have
>
> ```
> ├── office
>          ├── results
>                   ├── depth000000.png [...]

Oh, I found it. Truly, I am extremely grateful. This is indeed my own problem. I didn't check each file folder carefully. Thanks for the reminder. Sincerely thanking you again