luigifreda/pyslam #157: Relocalization Failed

Yasintrkmn, 2025-03-26:
When I use `main_slam.py` with LightGlue and DBOW3_INDEPENDENT or NETVLAD on KITTI 00,  I got "Relocalization Failed!'' constantly.
Why does that happen and can I do something to to solve?

luigifreda, 2025-03-26:
Hi,
Thanks for your feedback. This raises an excellent and challenging question: "Why do these advanced deep-learning methods struggle in this scenario?" I don’t have a definitive answer yet. I'm still in the process of fully understanding it. However, I can share a few observations.

First, in this specific case, the tracking failure stems from the **frontend**, rather than the relocalization/loop-closure method in the backend. When a SLAM system operates in exploration sequences (i.e., where the camera does not revisit mapped locations), tracking losses become critical if the camera quickly moves too far from mapped keyframes.

More broadly, I believe:
- All local features have inherent limitations. There is no "invincible" solution.
- Through testing, I found that some local features perform well for image matching and place recognition but struggle with feature tracking in SLAM. Since SLAM tightly integrates matching, geometry, and state estimation, additional geometric constraints can challenge the accuracy and robustness of local feature matches at both frame-to-frame and map-to-frame stages.
- Parameter tuning plays a crucial role. For instance, the descriptor matching ratio for outlier rejection should be adjusted for each specific local descriptor. For example, 0.7 is not universally optimal and requires fine-tuning. A better per-local-feature fine-tuning of the parameters is still a work in progress.

Regarding our case with KITTI 00, the issue is not with the relocalization method but with the local features used in the frontend. A few tests confirm this:
- Increasing SUPERPOINT’s matching ratio to 0.9 allows tracking to continue beyond frame 266, where it otherwise fails by default (at least this is what I observe on my machine).
- LIGHTGLUE_ALIKED appears more stable than the default LIGHTGLUE, which uses SUPERPOINT.
- Using `type: KITTI_DATASET` with color images improves tracking stability with LIGHTGLUE/SUPERPOINT, though failures still occur. Video compression and grayscale formats can lead to information loss and weaken some local features.

That said, I believe there's room for improvement, and pySLAM can likely be enhanced to better handle the diverse range of local features available. Additionally, we should take inspiration from emerging 3D-grounded methods like the promising XSt3R approaches.

I’d love to hear any further suggestions or observations you and other people might have.

Luigi

Yasintrkmn, 2025-03-26:
Thanks for detailed response.

closed 2025-03-26 by Yasintrkmn