luigifreda/pyslam #158: High RMSE on KITTI

Yasintrkmn, 2025-03-26:
I tested `main_slam.py` on KITTI 00 sequence. The ATE RMSE is 15.93.
Is it the expected result or could be something that I apply incorrectly? 

[image]

luigifreda, 2025-03-26:
Hi,
Thanks for your feedback.
Could you share your settings? I assume you're using the mono configuration with VIDEO, right?

Yasintrkmn, 2025-03-26:
Yes, my config is mono with video. I did not change anything else after pushing the repo.

luigifreda, 2025-03-26:
Another question. Are you evaluating the final or the online trajectory?

Yasintrkmn, 2025-03-26:
Final trajectory

luigifreda, 2025-03-26:
Monocular SLAM is challenging, especially on large-scale datasets where scale drift can become significant.

Below are the results I obtained on the same dataset.
Please note that pySLAM is not deterministic unless multithreading is disabled (though I still need to double-check this).

I ran a few tests with ORB-SLAM3 (monocular, same dataset), registering the monocular estimated trajectory (keyframes only) to the ground truth. The resulting RMSE/ATE remains within the same range. Please make sure to use the ground truth as the reference, not the other way around.

```
 EVO stats: {
    "rmse": 9.309509930718052,
    "mean": 8.73890223546111,
    "median": 8.519128309518509,
    "std": 3.209137402666154,
    "min": 2.3317783755723913,
    "max": 21.237804674955076,
    "sse": 391301.3928028733
}
```
[image]

closed 2025-03-26 by luigifreda

curious-energy, 2025-05-21:
First, thank you for your contribution. 

I pulled the latest GitHub repository, but running main_slam.py produced unsatisfactory results. What could be the issue with the settings? I tested two KITTI datasets, and the RMSE errors were both > 40m, with no configuration changes made. The image below shows the results of running the KITTI 00 dataset (monocular).
[image]

luigifreda, 2025-05-21:
Thanks for your feedback. It appears that loop-closing wasn't triggered in your case.
I reran the test using the latest version from the master branch and obtained a map similar to the one I shared [above](https://github.com/luigifreda/pyslam/issues/158#issuecomment-2754666593).
I'd be curious to know whether this behavior is consistent on your end or if it was a one-time occurrence.

curious-energy, 2025-05-22:
Thank you for your quick reply. Currently, this issue is consistently occurring on my end. I recently tried it on another computer of mine, and the problem of not being able to trigger the closed loop also appeared. I also believe the closed loop is not functioning at this point, but I haven't found any issues in all the configurations.

curious-energy, 2025-05-22:
I think in my reproduction, the GBA does not work. And I don't see any print information about GlobalBundleAdjustment. 

[image]
[image]

GBA logs are only one line.
[image]

luigifreda, 2025-05-22:
Thanks for confirming my hypothesis.
Could you please check whether you're seeing any RED errors in the main terminal or exceptions in the logs?

Which local features are you using?
If you're not using the default ORB, please refer to the README section on loop closure.
If you are using ORB, could you double-check that the vocabulary was properly downloaded?

Also, what OS and system configuration are you using? And are you connected to the internet while running the tests?

curious-energy, 2025-05-22:
During a large number of test runs, I only very occasionally saw one or two red errors related to tracking failures.

But apart from that, no other errors were found.  
However, I confirmed that both GBA and closing loop are in the True state.

I am using the combination of ORB2 and DBoW3.

I encountered network issues, but I placed the downloaded `orbvoc.dbow3` in the data path and confirmed that the path and name matched those in the code. Throughout the testing process, it was unable to connect to Google normally.

The systems I tested on different platforms were all Ubuntu 20.04.

I checked the initialization of DBoW3, and the issue is that even the nearby print statements couldn't be printed. I'm not sure why this is happening. I used the provided test script to confirm that DBoW3 can run normally.

luigifreda, 2025-05-24:
Hi, 
Could you please remove the vocabulary you manually placed there and let the code download it for you? 
Could you double-check you get something like this?

```bash
VocabularyData: downloading vocabulary ORB2 from: https://drive.google.com/uc?id=13xmRtop_ow3aPtv3qCT5beG19_mlogqI
downloading https://drive.google.com/uc?id=13xmRtop_ow3aPtv3qCT5beG19_mlogqI to /home/luigi/Work/pyslam/loop_closing/../data/ORBvoc.dbow3
Downloading...
From (original): https://drive.google.com/uc?id=13xmRtop_ow3aPtv3qCT5beG19_mlogqI
From (redirected): https://drive.google.com/uc?id=13xmRtop_ow3aPtv3qCT5beG19_mlogqI&confirm=t&uuid=58f7e4b9-d6f5-489c-9e05-1809cc4fb952
To: /home/luigi/Work/pyslam/data/ORBvoc.dbow3
 67%|██████████████████████████████████████████████████████████████████████████████████████████████████                                                 | 70.3M/105M [00:00<00:00, 80.5MB/s]Mplot2d: starting the process on figure: 1
Mplot2d: backend Qt5Agg
 75%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████▋                                     | 78.6M/105M [00:01<00:00, 77.1MB/s]Mplot2d: starting the process on figure: 2
Mplot2d: backend Qt5Agg
Mplot2d: starting the process on figure: 3
Mplot2d: backend Qt5Agg
Mplot2d: starting the process on figure: 4
Mplot2d: backend Qt5Agg
Viewer3D: Groundtruth shape: (1099, 3)
..................................
 82%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████▋                          | 86.5M/105M [00:01<00:00, 71.4MB/s]image: 0, timestamp: 0.0, duration: 0.1044989
 @tracking MONOCULAR, img id: 0, frame id: 0, state: NO_IMAGES_YET
img.shape: (370, 1226, 3), camera: 370x1226
timestamp:  0.0
detector: ORB2 , #features: 4000
descriptor: ORB2 , #features: 4000
detector: ORB2 , descriptor: ORB2 , #features: 4000  (#ref: 4000 ), [kp-filter: NONE ]
 94%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████▍         | 98.6M/105M [00:01<00:00, 83.7MB/s]..................................
image: 1, timestamp: 0.1044989, duration: 0.10329459999999999
 @tracking MONOCULAR, img id: 1, frame id: 1, state: NOT_INITIALIZED
img.shape: (370, 1226, 3), camera: 370x1226
timestamp:  0.1044989
detector: ORB2 , #features: 4000
descriptor: ORB2 , #features: 4000
detector: ORB2 , descriptor: ORB2 , #features: 4000  (#ref: 4000 ), [kp-filter: NONE ]
100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 105M/105M [00:01<00:00, 78.3MB/s]
|------------
Initializer: processing frames: 1, 0

```

luigifreda, 2025-05-24:
Could you please run `test/loopclosing/test_loop_detector.py` with `loop_detection_config = LoopDetectorConfigs.DBOW3` and tell me what you get?

curious-energy, 2025-05-24:
Thank you for your detailed response. I have resolved the issue, and PySLAM is now running as shown in the image you provided.

Here is a summary of the problems:  
1. Due to network issues, I manually placed the ORBvoc.dbow3 file, which was directly copied from the original DBoW3 repository. This caused the loop closure detection program to fail to start properly. After downloading the correct vocabulary file from the Google Drive link, the program successfully ran `test/loopclosing/test_loop_detector.py` with `loop_detection_config = LoopDetectorConfigs.DBOW3`. A clear indication that PySLAM is running correctly is the appearance of two windows named "loop closing," which do not appear (or appear as black windows) when the DBoW3 configuration is incorrect.  

2. At this point, the loop closure detection thread could be entered, but I still encountered issues with GBA (Global Bundle Adjustment). Upon checking the logs, I found an EXCEPTION message in `logs/loop_closing.log`, indicating a problem when calling g2o. It turned out that g2o was not correctly patched with the `git apply` command before compilation, leading to failed attribute calls during execution.  

Once again, thank you for your contributions. PySLAM is an excellent tool for experimenting with various deep learning methods.

```
EVO stats: {
    "rmse": 11.90034932063225,
    "mean": 9.781584756562326,
    "median": 7.212239055643163,
    "std": 6.777825138144241,
    "min": 0.7629747931775243,
    "max": 26.903594922605084,
    "sse": 642380.6720911367
}
```
[image]

JiePaiWs, 2025-09-08:
according to loop_closing.log, i find the issue:
Problem: Loop Closing thread starts successfully but never detects any loop candidates.
Log Analysis:
✅ Loop Closing thread starts normally
✅ Keyframes are properly received from LocalMapping
✅ First 10 keyframes skipped as expected (close to start)
❌ From keyframe 20 onwards: consistently "no loop candidates detected"
❌ LoopDetectingProcess returns empty candidate lists
❌ Subsequent stages (consistency check, geometry verification, GBA) never execute
Root Cause: The issue is stuck at the first stage of candidate detection in LoopDetectingProcess, preventing the entire loop closing pipeline from proceeding.

luigifreda, 2025-09-08:
Thanks. Do you mind attaching the full log files and the main terminal log?

JiePaiWs, 2025-09-09:
First of all, thank you for your work and reply. I finally identified the issue and resolved it.

[image]

    1.According to the logs, the loop_closing module couldn't even proceed through its initial steps properly. Due to my computer's relatively low configuration, I noticed that CV visualization was consuming significant resources. The Camera visualization window showed as unresponsive while other displays worked normally. Since loop_closing also uses CV to display related image information, I believe the lag in the Camera window was preventing subsequent processes from running correctly.

    2.After switching to --headless mode, loop_closing was able to proceed normally, and the final results were successful. Additionally, in **main_slam.py**,**line 332-336** .I commented out the following section, and loop closure also worked properly:

if not args.headless:
    img_draw = slam.map.draw_feature_trails(img)
    img_writer.write(img_draw, f"id: {img_id}", (30, 30))
    # 2D display (image display)
    cv2.imshow("Camera", img_draw)

Thank you again for your work. I would also suggest that if others encounter issues with Loop_closing not working when trying PYSLAM, they can use the same approach to **troubleshoot**. The issue is hard to resolve ,haha.