luigifreda/pyslam #159: an error of "cannot import name 'gtsam' from partially initialized module 'gtsam' " after excuting ./main_vo.py

byran-wang, 2025-03-27:
Hi @luigifreda :

Thanks for your great work.

 I get the an error of "cannot import name 'gtsam' from partially initialized module 'gtsam'" after excuting ./main_vo.py.

The detail logs are shown as following, can you help me to debug the error?

```
simba@simba4090:~/Documents/project/pyslam$ . pyenv-activate.sh
Activating pySLAM environment by using conda 
Preloading libffi...
(pyslam) simba@simba4090:~/Documents/project/pyslam$ ./main_vo.py
2025-03-27 15:56:39.664811: I tensorflow/core/util/port.cc:153] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2025-03-27 15:56:39.675707: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:467] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
E0000 00:00:1743062199.692980 2603371 cuda_dnn.cc:8579] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
E0000 00:00:1743062199.698403 2603371 cuda_blas.cc:1407] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
W0000 00:00:1743062199.711584 2603371 computation_placer.cc:177] computation placer already registered. Please check linkage and avoid linking the same target more than once.
W0000 00:00:1743062199.711646 2603371 computation_placer.cc:177] computation placer already registered. Please check linkage and avoid linking the same target more than once.
W0000 00:00:1743062199.711651 2603371 computation_placer.cc:177] computation placer already registered. Please check linkage and avoid linking the same target more than once.
W0000 00:00:1743062199.711653 2603371 computation_placer.cc:177] computation placer already registered. Please check linkage and avoid linking the same target more than once.
2025-03-27 15:56:39.715552: I tensorflow/core/platform/cpu_feature_guard.cc:210] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX_VNNI FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
 INFO: rerun running
Traceback (most recent call last):
  File "/home/simba/Documents/project/pyslam/./main_vo.py", line 64, in <module>
    from viewer3D import Viewer3D
  File "/home/simba/Documents/project/pyslam/viz/viewer3D.py", line 36, in <module>
    from map import Map
  File "/home/simba/Documents/project/pyslam/slam/map.py", line 45, in <module>
    import optimizer_gtsam
  File "/home/simba/Documents/project/pyslam/slam/optimizer_gtsam.py", line 30, in <module>
    import gtsam
  File "/home/simba/anaconda3/envs/pyslam/lib/python3.10/site-packages/gtsam/__init__.py", line 7, in <module>
    from gtsam import gtsam, utils
ImportError: cannot import name 'gtsam' from partially initialized module 'gtsam' (most likely due to a circular import) (/home/simba/anaconda3/envs/pyslam/lib/python3.10/site-packages/gtsam/__init__.py)
```

luigifreda, 2025-03-27:
Hi,
Thanks for your feedback. 
I’m unable to reproduce your issue. Could you let me know which OS you're using? I see you're using Conda. I tried reinstalling everything from scratch within a Conda environment, and it runs smoothly on my end.
Could you please double-check the following?
- You have pulled the latest version.
- You have built and installed GTSAM (ensure that thirdparty/gtsam/install exists).
- You have built gtsam_factors.

For the last two points, you can refer to the instructions here:
https://github.com/luigifreda/pyslam/tree/master/thirdparty/gtsam_factors 

If you have further issues, I suggest to git pull, clean and try a new install process. From pySLAM root folder:
```
git pull origin master 
./clean.sh --hard
-/pyenv-delete.sh
./install_all.sh
```

byran-wang, 2025-03-27:
I restalled the env, now the code can work. Thanks. @luigifreda