luigifreda/pyslam #166: ModuleNotFoundError: No module named 'depth_pro'

1443433629, 2025-04-09:
When I run main_slam.py, I found there is no depth_pro in the project. What should I do to solve this problem.

luigifreda, 2025-04-09:
Hi,
Thanks for your feedback. Can you provide some [details](https://github.com/luigifreda/pyslam/blob/master/docs/TROUBLESHOOTING.md#submitting-a-git-issue) about your OS, CUDA version, install method, etc.?

1443433629, 2025-04-10:
> Hi, Thanks for your feedback! Could you please share some [details](https://github.com/luigifreda/pyslam/blob/master/docs/TROUBLESHOOTING.md#submitting-a-git-issue) about your setup - such as your OS, CUDA version, and installation method?
> 
> Also, did you run `. pyenv-activate.sh`? (Please double-check that there's a space after the dot.)
> 
> Thanks!
> 
> P.S.: This morning, I ran a fresh install from scratch under Docker, and it worked fine.

I'm running this project on WSL2 Ubuntu 20.04. The CUDA version I'm using is 24.1.2. I followed the method in the README and ran ./install_all.sh. After the installation was finished, I ran ./main_vo.py and it worked. But when I ran ./main_slam.py, it threw this error:
Traceback (most recent call last):
  File "/home/zzj/pyslam/./main_slam.py", line 32, in <module>
    from slam import Slam, SlamState
  File "/home/zzj/pyslam/slam/slam.py", line 57, in <module>
    from volumetric_integrator_base import VolumetricIntegrationOutput
  File "/home/zzj/pyslam/dense/volumetric_integrator_base.py", line 50, in <module>
    from depth_estimator_factory import depth_estimator_factory, DepthEstimatorType
  File "/home/zzj/pyslam/depth_estimation/depth_estimator_factory.py", line 37, in <module>
    from depth_estimator_depth_anything_v2 import DepthEstimatorDepthAnythingV2
  File "/home/zzj/pyslam/depth_estimation/depth_estimator_depth_anything_v2.py", line 41, in <module>
    from depth_anything_v2.dpt import DepthAnythingV2
ModuleNotFoundError: No module named 'depth_anything_v2'

PS: Yesterday, I imported the DepthPro source code into the project, and there were no more errors related to DepthPro. But I feel it's too much trouble to import things into the project one by one.

Today, I followed your suggestion and ran . pyenv-activate.sh. But I got this error: -bash: /home/zzj/anaconda3/envs/pyslam/bin/activate: No such file or directory.

1443433629, 2025-04-11:
> Hi, I strongly recommend using a native Ubuntu installation. At the moment, I'm unable to verify any WSL2-related configurations on Windows.
> 
> To properly diagnose the issue, I need the complete output log from the `install_all.sh` script. You shouldn’t need to download anything manually - the script is designed to handle all dependencies automatically.
> 
> Please share the full log, as it's likely that multiple errors are occurring and interrupting the normal installation process.
> 
> From the output of `pyenv-activate.sh`, it looks like the script isn't detecting Conda and is defaulting to `venv`. Since your system already has Conda installed, could you try following this procedure instead? 👉 https://github.com/luigifreda/pyslam/blob/master/docs/CONDA.md
> 
> Let me know how it goes.

Thanks so much for your help! I realized earlier I’d been running install_all_conda.sh instead of install_all.sh. Today I ran install_all.sh and then ./main_slam.py, and it worked perfectly just as I’d hoped!
Thank you again for this amazing project! It turns out it runs smoothly on WSL2 too, and gives the visual outputs we wanted. That’s really impressive!

closed 2025-04-11 by 1443433629