luigifreda/pyslam #170: How to improve trajectory without sudden drifts

shashankyld, 2025-04-20:
I ran the main_slam.py with default params and I am unable to figure out why in somecases tracking fails as show in the figure.Sudden drop in the matched map_points in the middle of the sequence. Please help me come up with params that are suitable for the most stable tracking independent of the speed. 

[image]

My current slam thread params
```

    # SLAM threads 
    kLocalMappingOnSeparateThread=True               # True: move local mapping on a separate thread, False: tracking and then local mapping in a single thread 
    kTrackingWaitForLocalMappingToGetIdle=False      # True: wait for local mapping to be idle before starting tracking, False: tracking and then local mapping in a single thread  
    kTrackingWaitForLocalMappingSleepTime=0.1        # DEPRECATED: -1 for no sleep # [s]    (NOTE: a bit of sleep time increases the call rate of LBA and therefore VO accuracy)
    kLocalMappingParallelKpsMatching=True
    kLocalMappingParallelKpsMatchingNumWorkers=6
    kLocalMappingDebugAndPrintToFile = True
```

Some additional log that might be useful
```
failure in tracking previous frame, # matched map points: 4
>>>> tracking reference 164 ...
matching keypoints with  BF
# keypoints matched: 25 
descriptor sigma:  115.28017144979967
checking orientation consistency - valid matches % : 60.0 % of  25 matches
# matched map points in prev frame: 15 
pose opt match-frame-keyframe 
pose optimization: available 63 points, found 20 bad points
     error^2: 2.998653,  ok: 1
 failure in tracking reference 164, # matched map points: 4
 tracking failure

```

I will keep trying with different sequences and post here in case i find similar tracking failures in other sequences

luigifreda, 2025-04-20:
Thanks for your feedback!

When reporting test results, it would be very helpful to include the full log and the dataset name.

If I'm not mistaken, this dataset is TUM FrDesk 1, which includes sequences with direction changes and motion blur. I noticed the tracking loss compared to ORB-SLAM's performance - this is a known issue, and some possible causes are outlined in the troubleshooting section:
🔗 [Troubleshooting - Bad Tracking Performances](https://github.com/luigifreda/pyslam/blob/master/docs/TROUBLESHOOTING.md#bad-tracking-performances)

As a workaround, try setting `kLocalMappingOnSeparateThread=False`. This often prevents the tracker from getting lost in these sequences, where some performance differences are observed.

That said, I’ve also observed some regressions introduced by recent changes. Performance improvements for PySLAM are currently a work in progress.

luigifreda, 2025-05-08:
Just pushed a couple of improvements that improve performance on TUM FrDesk 1 as well.

luigifreda, 2025-07-16:
https://github.com/luigifreda/pyslam/tree/tracking_improvements

luigifreda, 2025-07-19:
Hi,
Thank you for your feedback. Just to clarify, this test did not involve any relocalization or loop closure.

https://github.com/luigifreda/pyslam/issues/170#issuecomment-3079170852

luigifreda, 2025-07-19:
These are two runs I got on the current `master` where I explicitly set  `kUseLoopClosing = False`. 

[image]
[image]

`

luigifreda, 2025-07-20:
Hi @JiePaiWs,
Thanks again for your feedback.

I’ve run the tests on the current master branch without modifying any parameters.

If you need support:
- Make sure your local copy is up to date by pulling the latest changes from the master branch. Then perform a clean reset and re-install pySLAM, as described [here](https://github.com/luigifreda/pyslam/blob/master/docs/TROUBLESHOOTING.md#clean-reset).
- I highly recommend including the [suggested information](https://github.com/luigifreda/pyslam/blob/master/docs/TROUBLESHOOTING.md#submitting-a-git-issue) when reporting issues.

In particular, please share the same types of plots and viewer screenshots I’ve posted (e.g. [this example](https://github.com/luigifreda/pyslam/issues/170#issuecomment-3092399183)), along with the full console log.

JiePaiWs, 2025-07-25:
> Hi, Thanks for your notes. OpenCV version is not a problem here. Please, try the new version 2.8.7. I pushed some optimizations and bug fixes. LMK

Thank you for your reply.I try the new version 2.8.7. in Docker ,but the tracking performance is still poor,I set the   
* kLocalMappingOnSeparateThread = False            
* kTrackingWaitForLocalMappingToGetIdle = False 
I wonder why the tracking performance is poor in my PC,could you give me some advice ?Looking forward to your reply.
Some additional log that might be useful:
'''
F(395) #matched points: 370, KF(393) #matched points: 516
is_local_mapping_idle:  True , local_mapping_queue_size:  0
 NEW KF
 Adding new KF with id 395, img shape: (480, 640, 3), img_right shape: None, depth shape: (480, 640)
New keyframe created: 395, local_mapping_queue_size: 1
local mapping: starting...
 @local mapping
>>>> updating map points (1000)...
>>>> updating connections ...
>>>> culling map points...
#culled points: 449, timing: 0.001626949
>>>> creating new map points
local map keyframes:  [393, 391, 390, 389, 388, 385, 384, 382, 383, 380]  +  395 ...
compute_frame_matches: #compared pairs: 10, do_parallel: True, timing: 0.032189414
	 #added map points: 6 for KFs (395), (389)
	 #added map points: 5 for KFs (395), (388)
	 #added map points: 3 for KFs (395), (385)
	 #added map points: 0 for KFs (395), (384)
	 #added map points: 1 for KFs (395), (382)
#new map points: 15, timing: 0.036490187
>>>> fusing map points
local map keyframes:  [384, 385, 388, 389, 390, 391, 393, 377, 378, 379, 380, 382, 383]  +  395 ...
	 #fused map points: 179 for KFs (395, 384)
	 #fused map points: 113 for KFs (395, 385)
	 #fused map points: 99 for KFs (395, 388)
	 #fused map points: 74 for KFs (395, 389)
	 #fused map points: 53 for KFs (395, 390)
	 #fused map points: 42 for KFs (395, 391)
	 #fused map points: 56 for KFs (395, 393)
	 #fused map points: 16 for KFs (395, 377)
	 #fused map points: 15 for KFs (395, 378)
	 #fused map points: 17 for KFs (395, 379)
	 #fused map points: 6 for KFs (395, 380)
	 #fused map points: 15 for KFs (395, 382)
	 #fused map points: 7 for KFs (395, 383)
	 #fused map points: 31 for local map into KF 395
#fused map points: 723, timing: 0.29071679
>>>> local optimization (LBA) ...
local optimization window:  [296, 297, 298, 300, 301, 302, 303, 304, 305, 306, 307, 308, 311, 313, 315, 326, 330, 333, 334, 336, 337, 338, 339, 340, 341, 342, 347, 348, 349, 350, 351, 352, 353, 354, 355, 356, 357, 358, 359, 360, 361, 362, 363, 364, 365, 366, 367, 368, 370, 371, 372, 373, 375, 377, 378, 379, 380, 382, 383, 384, 385, 388, 389, 390, 391, 393, 395]
                     refs:  [0, 1, 2, 4, 5, 6, 8, 9, 10, 11, 12, 14, 15, 16, 17, 18, 19, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 32, 35, 36, 37, 38, 39, 40, 89, 90, 91, 97, 110, 112, 115, 117, 118, 119, 121, 123, 124, 125, 126, 127, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 144, 145, 146, 147, 148, 149, 150, 151, 152, 153, 155, 156, 157, 158, 159, 160, 161, 162, 163, 166, 167, 168, 169, 170, 171, 172, 180, 183, 184, 185, 186, 187, 188, 190, 191, 192, 193, 194, 195, 196, 199, 204, 205, 214, 215, 219, 224, 225, 228, 229, 230, 231, 232, 233, 235, 236, 237, 238, 240, 241, 245, 255, 256, 259, 260, 262, 263, 265, 267, 268, 271, 272, 273, 275, 282, 284, 285, 287, 288, 289, 292, 293, 314, 316, 317, 318, 319, 320, 321, 322, 323, 324, 325, 327, 328, 329]
                   #points:  4862
local_bundle_adjustment: starting optimization with 226 keyframes, 4862 points, 41008 edges
 local optimization - perc bad observations: 0.41 %
local optimization (LBA) error^2: 1.7660414783268825, timing: 8.694233046
 KF(395) #points: 490 
>>>> culling keyframes...
	 #culled keyframes: 0, timing: 0.272264182
local mapping elapsed time: 9.354445457458496
 map: 12798 points, 247 keyframes
Tracking: elapsed_time:  9.833909034729004
..................................

'''

luigifreda, 2025-07-25:
Hi,
Thanks for your feedback and for running the tests.

To better assist you, could you please share the full log as well as snapshots of the 3D viewer and other plots (similar to the ones above)? Without this information, it's quite difficult to diagnose the issue effectively.

Just to confirm, are you running on commit 8e2b0d697381b087ab5034ba681569c9d3499b80 (version 2.8.7), and did you reinstall everything from scratch after performing a clean reset as described in the [troubleshooting guide](https://github.com/luigifreda/pyslam/blob/master/docs/TROUBLESHOOTING.md#clean-reset)?

Regarding the potential performance issue: I'm currently using OpenCV 4.5.5 due to known conflicts between OpenCV 4.10 and some libraries in Ubuntu 20.04. Could this be the root cause?

As far as I know, you’re working within a Docker environment. Could you share the corresponding Dockerfile? Also, have you considered trying [pyenv](https://github.com/pyenv/pyenv) or pixi instead of conda?

Lastly, could you clarify what specific issues you're encountering with the OpenCV installation? It would also be helpful if you could share the full output from running [./scripts/install_opencv_python.sh](https://github.com/luigifreda/pyslam/blob/master/scripts/install_opencv_python.sh) within the pyslam Python environment.

luigifreda, 2025-07-25:
New version 4.8.8 available.