luigifreda/pyslam #171: a trouble when running

yangyifei1024, 2025-04-26:
Hello!
After finishing ./install_all.sh I started main_slam.py and an error occur:

[image]

ModuleNotFoundError: No module named 'pnpsolver

Then  I turn to main_vo.py and another issue:

[image]
ModuleNotFoundError:No module named 'glutils

Have you encountered thisissue before? Could you please tell mehow to install these two packages:pnpsolver and glutils? lf it's convenientfor you, could you help me solve this?Thank you very much!

dvdmc, 2025-04-26:
Hi @yangyifei1024 , some additional info could be useful.
Which OS are you running?
Are you using Conda or Pyenv installation?
Do you see any error during the installation process?
The problem is likely that there was a problem during the installation that prevented the script from installing those libraries.

shashankyld, 2025-04-26:
Try running `./install_cpp.sh`

luigifreda, 2025-04-27:
Hi everyone,

Thank you so much to @dvdmc and @shashankyld for supporting @yangyifei1024!

Yes, as @dvdmc pointed out, it would be very helpful to share all the relevant information following the troubleshooting guidelines [here](https://github.com/luigifreda/pyslam/blob/master/docs/TROUBLESHOOTING.md).
This information is crucial for reproducing and diagnosing the issue.

It seems that the Python bindings in the cpp folder were not built correctly. As @shashankyld suggested, you should verify the build using the install_cpp.sh script. Before running it, make sure to activate your Python environment. Here's what you should do:

Open a new terminal and navigate to your pyslam working copy folder.

Run the following commands:
```bash
. pyenv-activate.sh
./install_cpp.sh
```
You might encounter some build errors. If so, please share the full build log so we can help you further.

Thanks again for your collaboration!

yangyifei1024, 2025-04-28:
@luigifreda @dvdmc @shashankyld Thank you very much for your reply!  I have tried your suggestion

`./install_cpp.sh`

The following error message was generated:

================================================ 
Building and installing cpp ... 
Currently used pyenv: pyslam 
EXTERNAL_OPTIONS: 
-- found OpenCV version: 4.5.4
-- opencv include: /usr/include/opencv4
-- opencv lib dirs: /usr
-- opencv libs: opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_stitching;opencv_video;opencv_videoio;opencv_alphamat;opencv_aruco;opencv_barcode;opencv_bgsegm;opencv_bioinspired;opencv_ccalib;opencv_datasets;opencv_dnn_objdetect;opencv_dnn_superres;opencv_dpm;opencv_face;opencv_freetype;opencv_fuzzy;opencv_hdf;opencv_hfs;opencv_img_hash;opencv_intensity_transform;opencv_line_descriptor;opencv_mcc;opencv_optflow;opencv_phase_unwrapping;opencv_plot;opencv_quality;opencv_rapid;opencv_reg;opencv_rgbd;opencv_saliency;opencv_shape;opencv_stereo;opencv_structured_light;opencv_superres;opencv_surface_matching;opencv_text;opencv_tracking;opencv_videostab;opencv_viz;opencv_wechat_qrcode;opencv_ximgproc;opencv_xobjdetect;opencv_xphoto
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindOpenGL.cmake:315 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  CMakeLists.txt:27 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- WITH_PYTHON_INTERP_CHECK: ON
-- PythonInterp: 
CMake Error at CMakeLists.txt:56 (add_subdirectory):
  add_subdirectory given source "pybind11" which is not an existing
  directory.

CMake Error at CMakeLists.txt:58 (pybind11_add_module):
  Unknown CMake command "pybind11_add_module".

-- Configuring incomplete, errors occurred!
See also "/codes/yyf/pyslam/cpp/build/CMakeFiles/CMakeOutput.log".

Do you know how to solve it? I am looking forward to your answer.

luigifreda, 2025-04-29:
Hi,
Thanks for sharing the log.
It looks like there is a missing symbolic link `pybind11 -> ../thirdparty/pybind1` in the `cpp` folder, or the `thirdparty/pybind11` folder itself is missing. This is unexpected. Could you please double-check if you can successfully run `cd cpp/pybind11`?
Also, could you let me know which operating system you are using?

closed 2025-05-02 by luigifreda