luigifreda/pyslam #178: Adding MegaLoc as global feature extractor?

gmberton, 2025-06-02:
I'm wondering if it might be useful for this repo to add MegaLoc, which is the latest SOTA global features method in Visual Place Recognition.
It is much more powerful than models like CosPlace and EigenPlaces (achieves SOTA on almost any VPR dataset, Visual Localization and Landmark Retrieval), but it is also quite heavier (228M parameters).

The nice thing is that the weights are on torch.hub (like EigenPlaces) so the model can be used just like
`model = torch.hub.load("gmberton/MegaLoc", "get_trained_model")`

Link to the paper [https://arxiv.org/pdf/2502.17237v1](https://arxiv.org/pdf/2502.17237v1) and to the code [https://github.com/gmberton/MegaLoc](https://github.com/gmberton/MegaLoc)

I'd be happy to help if needed

And thanks for the amazing repo!

luigifreda, 2025-06-02:
Hi Gabriele,
Thank you so much for your feedback and for sharing your fantastic projects.

I’ve just integrated MegaLoc into pySLAM, following the same structure as CosPlace and EigenPlaces. The integration was smooth and straightforward. I’ve pushed the changes. Feel free to take a look and double-check that everything is being used correctly:
[test_vpr.py](https://github.com/luigifreda/pyslam/blob/master/test/loopclosing/test_vpr.py)
[global_feature_megaloc.py](https://github.com/luigifreda/pyslam/blob/master/loop_closing/global_feature_megaloc.py)

Thanks again for your support and for the amazing work you’re doing.
Best regards,
Luigi

closed 2025-06-02 by luigifreda

gmberton, 2025-06-02:
The code looks great, and thank you for the amazing repo and the huge work maintaining it!

luigifreda, 2025-06-02:
Grazie Gabriele. The same goes for you and your awesome repositories :-)