luigifreda/pyslam #179: 2D feature point in pixel coordinate and 3D point in map

laiyuzhi, 2025-06-02:
Hi, Sorry to bother you again. Similar to #176 , could you point out how I can get the keypoints that are matched in each frame and their 3D location in the map? （2D feature point in pixel coordinate and 3D point in map）

luigifreda, 2025-06-02:
Hi,
You can find this information in the `Frame` class.
Keypoints: https://github.com/luigifreda/pyslam/blob/37a6ded2c38d95c794890ab3214fd8a22edc4636/slam/frame.py#L367
Mapoints: https://github.com/luigifreda/pyslam/blob/37a6ded2c38d95c794890ab3214fd8a22edc4636/slam/frame.py#L382

Hope this helps.

closed 2025-06-02 by luigifreda

laiyuzhi, 2025-06-02:
Thank you for your help, may I ask what coordinate system this points is relative to?