luigifreda/pyslam #180: ImportError: /home/mons/miniconda3/envs/pyslam/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.32' not found

ufukasia, 2025-06-05:
dear Luigi Freda 
i getting this error on pySLAM v2.8.0. I am on ubuntu24. How can i solve this problem. 

(pyslam) (base) mons@mons-ABRA-A5-V13-5:~/Desktop/pyslam$ /home/mons/miniconda3/envs/pyslam/bin/python /home/mons/Desktop/pyslam/main_slam.py
2025-06-05 19:51:25.600698: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:467] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
E0000 00:00:1749142285.697861   95227 cuda_dnn.cc:8579] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
E0000 00:00:1749142285.732279   95227 cuda_blas.cc:1407] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
W0000 00:00:1749142285.963917   95227 computation_placer.cc:177] computation placer already registered. Please check linkage and avoid linking the same target more than once.
W0000 00:00:1749142285.963951   95227 computation_placer.cc:177] computation placer already registered. Please check linkage and avoid linking the same target more than once.
W0000 00:00:1749142285.963954   95227 computation_placer.cc:177] computation placer already registered. Please check linkage and avoid linking the same target more than once.
W0000 00:00:1749142285.963956   95227 computation_placer.cc:177] computation placer already registered. Please check linkage and avoid linking the same target more than once.
2025-06-05 19:51:25.989684: I tensorflow/core/platform/cpu_feature_guard.cc:210] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
/home/mons/miniconda3/envs/pyslam/lib/python3.10/site-packages/torch/cuda/__init__.py:141: UserWarning: CUDA initialization: Unexpected error from cudaGetDeviceCount(). Did you run some cuda functions before calling NumCudaDevices() that might have already set an error? Error 804: forward compatibility was attempted on non supported HW (Triggered internally at ../c10/cuda/CUDAFunctions.cpp:108.)
  return torch._C._cuda_getDeviceCount() > 0
Traceback (most recent call last):
  File "/home/mons/Desktop/pyslam/main_slam.py", line 19, in <module>
    from slam import Slam, SlamState
  File "/home/mons/Desktop/pyslam/slam/slam.py", line 31, in <module>
    from frame import Frame, FeatureTrackerShared, match_frames
  File "/home/mons/Desktop/pyslam/slam/frame.py", line 35, in <module>
    from camera_pose import CameraPose
  File "/home/mons/Desktop/pyslam/slam/camera_pose.py", line 21, in <module>
    import g2o
ImportError: /home/mons/miniconda3/envs/pyslam/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.32' not found (required by /home/mons/Desktop/pyslam/thirdparty/g2opy/lib/g2o.cpython-310-x86_64-linux-gnu.so)

ufukasia, 2025-06-06:
i am trying with new ubuntu 2.4 and clean pslam. when i do "conda install -c conda-forge libstdcxx-ng"
this is the new error 

Traceback (most recent call last):
  File "/home/mons/Desktop/pyslam/main_slam.py", line 400, in <module>
    cv2.imshow('Camera', img_draw)
cv2.error: OpenCV(4.10.0) /io/opencv/modules/highgui/src/window.cpp:1301: error: (-2:Unspecified error) The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Cocoa support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script in function 'cvShowImage'

luigifreda, 2025-06-08:
Hi, 
I've pushed new improvements to the install process on master.

closed 2025-06-11 by luigifreda