luigifreda/pyslam #183: Is there any configuration that can make GTSAM pose estimation more stable?

RazvanTrifuMovella, 2025-06-26:
Hi! I'm working on a project where I'm trying to extract pose orientation covariance, and I noticed that this is not easily available through g2o. Since the bindings don't allow me to access computeMarginals() from g2o. I've tried making my own bindings using pybind11, but I don't have a lot of experience with that and I was unsuccessful. 

GTSAM exposes covariance extraction more conveniently, so I tried switching to it for pose estimation. However, I re-localization often fails when using GTSAM in pySLAM (no new keyframes created, or tracking doesn't recover fully).

I've uploaded a demonstration here:
[https://imgur.com/a/bq1rCoQ](https://imgur.com/a/bq1rCoQ)

luigifreda, 2025-07-10:
Hi Razvan,
Thanks for your message and for sharing your workaround. I do see a few potential limitations in the current method:

1. Neglecting map uncertainty:  Your procedure assumes the rest of the graph (i.e., landmarks and other variables) is fixed, so it only captures the uncertainty of the current pose *conditioned* on a fixed map.  If you want the *marginal covariance* of the pose in the full graph, you'd need to marginalize over the other variables, typically using something like the Schur complement or full marginalization tools (e.g., sparse Cholesky or dense inverse of the Hessian from the full optimizer).

2. Missing measurement noise modeling: You're building an approximate information matrix as: `H = J.T @ J`. This implicitly assumes unit variance - i.e., all measurements have equal and isotropic noise.  
In practice, you should use: `H = J.T @ Sigma^-1 @ J` where `Sigma` is the block-diagonal matrix of measurement covariances. Ignoring this can bias the result if measurements have heterogeneous noise characteristics (e.g., stereo vs. mono, depth vs. RGB).

3. Perturbation model: The adopted perturbation is `SE3.exp(delta) * T`, which is a left perturbation.  The resulting covariance corresponds to uncertainties expressed in the *local (body) frame*.  That's totally valid, but it's worth clarifying: you're computing *local pose uncertainty in tangent space*, *not* the full global pose covariance.

That said, your approach reminds me of something close to the sigma point method for local pose uncertainty approximation.
As you know, for rigorous results, you'd want a more accurate marginalization method or a linearization around the full graph.

I'll take a look on how to expose `optimizer.computeMarginals()` ASAP. 

Best,
Luigi

RazvanTrifuMovella, 2025-07-11:
Hi Luigi,

Thank you for adding the binding to compute_Marginals()!

I’m calling it right after a successful pose optimization:

`    if is_ok:
        est = v_se3.estimate()
        R   = est.rotation().matrix()
        t   = est.translation()
        frame.update_pose(poseRt(R, t))
        ok, spinv = opt.compute_marginals(0)  
        print(ok) `

The optimisation converges, but ok is always False. Is there a recommended solver/ordering combo or an extra step I should run before using compute_marginals? 

Cheers,
Razvan