luigifreda/pyslam #193: CUDA version

hajun011103, 2025-08-12:
Hello, first of all, thank you for your work

im running on CUDA version 12.2 and i have matched the cuda toolkit as well.

meic@meic:~$ nvidia-smi
Tue Aug 12 10:12:53 2025       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.247.01             Driver Version: 535.247.01   CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 4060        Off | 00000000:01:00.0 Off |                  N/A |
|  0%   30C    P8              N/A / 115W |      8MiB /  8188MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A      2093      G   /usr/lib/xorg/Xorg                            4MiB |
+---------------------------------------------------------------------------------------+
meic@meic:~$ nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Tue_Jun_13_19:16:58_PDT_2023
Cuda compilation tools, release 12.2, V12.2.91
Build cuda_12.2.r12.2/compiler.32965470_0

But after i complete the installation with conda 
it misinstalled some pips such as torch==2.2.0 and so on.
futhermore, it didnt install some pips and libraries because there are not distributions which match the versions.

Does pyslam also support for cuda 12.2 as well?
If not, what versions does it support?

Thank you for your a lot of work on slam.
It is a huge help for me studying slam.

luigifreda, 2025-08-12:
Hi,
Thanks for your feedback. Unfortunately, I’m unable to clearly identify your exact problem beyond the nominal package requirements.
Could you please let me know:
- Your operating system
- The full error log you get when running main_slam.py
- The complete log from when you installed pyslam

Also, I’m curious, why did you change the name of the Python environment?

Without these details, it’s very difficult to provide effective support.

hajun011103, 2025-08-20:
Sorry for the late reply.

Im currently using Ubuntu 22.04.
I have resolved the problem with adding one line in the bashrc.
But I have no idea why it did not work and now it is working.

```shell
export LD_LIBRARY_PATH=/home/meic/anaconda3/envs/pyslam/lib:$LD_LIBRARY_PATH
``` 

However, I have another problem
I have disabled the depth estimation in the config_parameters.py
but ./main_slam.py did not work.
I have no idea what has been changed but it suddenly does not work anymore.

```shell
(pyslam) meic@meic:/mnt/new/pyslam$ ./main_slam.py 
2025-08-20 19:32:24.462167: I tensorflow/core/util/port.cc:153] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2025-08-20 19:32:24.472029: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:467] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
E0000 00:00:1755685944.485016   32167 cuda_dnn.cc:8579] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
E0000 00:00:1755685944.488329   32167 cuda_blas.cc:1407] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
W0000 00:00:1755685944.496457   32167 computation_placer.cc:177] computation placer already registered. Please check linkage and avoid linking the same target more than once.
W0000 00:00:1755685944.496468   32167 computation_placer.cc:177] computation placer already registered. Please check linkage and avoid linking the same target more than once.
W0000 00:00:1755685944.496470   32167 computation_placer.cc:177] computation placer already registered. Please check linkage and avoid linking the same target more than once.
W0000 00:00:1755685944.496472   32167 computation_placer.cc:177] computation placer already registered. Please check linkage and avoid linking the same target more than once.
2025-08-20 19:32:24.499020: I tensorflow/core/platform/cpu_feature_guard.cc:210] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX512F AVX512_VNNI AVX512_BF16 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
/home/meic/anaconda3/envs/pyslam/lib/python3.11/site-packages/f3rm/features/clip/clip.py:6: UserWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html. The pkg_resources package is slated for removal as early as 2025-11-30. Refrain from using this package or pin to Setuptools<81.
  from pkg_resources import packaging
/home/meic/anaconda3/envs/pyslam/lib/python3.11/site-packages/torch/cuda/__init__.py:141: UserWarning: CUDA initialization: CUDA unknown error - this may be due to an incorrectly set up environment, e.g. changing env variable CUDA_VISIBLE_DEVICES after program start. Setting the available devices to be zero. (Triggered internally at ../c10/cuda/CUDAFunctions.cpp:108.)
  return torch._C._cuda_getDeviceCount() > 0
DepthEstimatorCrestereoMegengine: megengine not found
Aborted (core dumped)
``` 

Thank you for your amazing work btw

and the pyslam_backup conda env is literally just backup env 
i have conda cloned with a name of pyslam_backup
i just did not want to destory the original conda env (pyslam)

luigifreda, 2025-08-20:
Can you check if this is the problem?
https://github.com/luigifreda/pyslam/blob/master/docs/TROUBLESHOOTING.md#aborted-core-dumped-or-segmentation-fault-core-dumped

luigifreda, 2025-08-21:
Above, you mentioned that you added 
```bash 
export LD_LIBRARY_PATH=/home/meic/anaconda3/envs/pyslam/lib:$LD_LIBRARY_PATH
```
That should NOT be needed at all if you run `. pyenv-activate.sh` (use the `.` separated from ` pyenv-activate.sh`) as reported in the install/usage instructions (see, for example, [here](https://github.com/luigifreda/pyslam?tab=readme-ov-file#usage)).
Indeed, under conda, the script `./scripts/pyenv-conda-activate.sh` is called by `pyenv-activate.sh` and the same export is run [here](https://github.com/luigifreda/pyslam/blob/master/scripts/pyenv-conda-activate.sh#L67).
Did you run `. pyenv-activate.sh` before running `main_slam.py`.

If the problem persists, you can gather more details by running your script under gdb:
```bash
gdb --args python main_slam.py
(gdb) run
# when it aborts:
(gdb) bt
```
In this case, please attach the logs (including the gdb backtrace), so we can investigate further.

luigifreda, 2025-08-21:
I succeeded in reproducing the problem using conda. 
On my side, the gdb bt returned
```term
Thread 40 "jemalloc_bg_thd" received signal SIGSEGV, Segmentation fault. [Switching to Thread 0x7ffdeedff700 (LWP 121820)] 0x00007ffdf108ec81 in background_thread_entry () from /media/ssd2/data/miniconda3/envs/pyslam/lib/python3.11/site-packages/pyarrow/libarrow.so.2100 (gdb) bt #0 0x00007ffdf108ec81 in background_thread_entry () from /media/ssd2/data/miniconda3/envs/pyslam/lib/python3.11/site-packages/pyarrow/libarrow.so.2100 #1 0x00007ffff7f4a609 in start_thread (arg=<optimized out>) at pthread_create.c:477 #2 0x00007ffff7d15353 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

I successfully fixed the problem by running 
```bash
pip install "pyarrow<19"
```

hajun011103, 2025-09-01:
Thank you so much for your reply

I have followed your instructions and I got this error. (in the same terminal)
I have found this kind of error but I could not resolve this.
I successfully run until 
```shell
./thirdparty/lietorch/build.sh
``` 

And this is what I got while downloading simple knn
```shell
(pyslam) meic@meic:/mnt/new/pyslam$ pip install ./thirdparty/monogs/submodules/simple-knn/
Processing ./thirdparty/monogs/submodules/simple-knn
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: simple_knn
  DEPRECATION: Building 'simple_knn' using the legacy setup.py bdist_wheel mechanism, which will be removed in a future version. pip 25.3 will enforce this behaviour change. A possible replacement is to use the standardized build interface by setting the `--use-pep517` option, (possibly combined with `--no-build-isolation`), or adding a `pyproject.toml` file to the source tree of 'simple_knn'. Discussion can be found at https://github.com/pypa/pip/issues/6334
  Building wheel for simple_knn (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [305 lines of output]
      g++ version: 11.4.0
      gcc major version: 11
      nvcc flags: ['-O2', '-allow-unsupported-compiler', '-std=c++20', '-gencode=arch=compute_50,code=sm_50', '-gencode=arch=compute_52,code=sm_52', '-gencode=arch=compute_53,code=sm_53', '-gencode=arch=compute_60,code=sm_60', '-gencode=arch=compute_61,code=sm_61', '-gencode=arch=compute_62,code=sm_62', '-gencode=arch=compute_70,code=sm_70', '-gencode=arch=compute_72,code=sm_72', '-gencode=arch=compute_75,code=sm_75', '-gencode=arch=compute_80,code=sm_80', '-gencode=arch=compute_86,code=sm_86', '-gencode=arch=compute_87,code=sm_87', '-gencode=arch=compute_89,code=sm_89', '-gencode=arch=compute_90,code=sm_90']
      cxx flags: ['-O2', '-std=c++20']
      running bdist_wheel
      running build
      running build_ext
      Building with 4 parallel jobs
      /home/meic/anaconda3/envs/pyslam/lib/python3.11/site-packages/torch/utils/cpp_extension.py:415: UserWarning: The detected CUDA version (12.2) has a minor version mismatch with the version that was used to compile PyTorch (12.1). Most likely this shouldn't be a problem.
        warnings.warn(CUDA_MISMATCH_WARN.format(cuda_str_version, torch.version.cuda))
      /home/meic/anaconda3/envs/pyslam/lib/python3.11/site-packages/torch/utils/cpp_extension.py:425: UserWarning: There are no g++ version bounds defined for CUDA version 12.2
        warnings.warn(f'There are no {compiler_name} version bounds defined for CUDA version {cuda_str_version}')
      building 'simple_knn._C' extension
      creating /mnt/new/pyslam/thirdparty/monogs/submodules/simple-knn/build/temp.linux-x86_64-cpython-311
      Emitting ninja build file /mnt/new/pyslam/thirdparty/monogs/submodules/simple-knn/build/temp.linux-x86_64-cpython-311/build.ninja...
      Compiling objects...
      Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
      [1/3] /usr/local/cuda-12.2/bin/nvcc --generate-dependencies-with-compile --dependency-output /mnt/new/pyslam/thirdparty/monogs/submodules/simple-knn/build/temp.linux-x86_64-cpython-311/simple_knn.o.d -I/home/meic/anaconda3/envs/pyslam/lib/python3.11/site-packages/torch/include -I/home/meic/anaconda3/envs/pyslam/lib/python3.11/site-packages/torch/include/torch/csrc/api/include -I/home/meic/anaconda3/envs/pyslam/lib/python3.11/site-packages/torch/include/TH -I/home/meic/anaconda3/envs/pyslam/lib/python3.11/site-packages/torch/include/THC -I/usr/local/cuda-12.2/include -I/home/meic/anaconda3/envs/pyslam/include/python3.11 -c -c /mnt/new/pyslam/thirdparty/monogs/submodules/simple-knn/simple_knn.cu -o /mnt/new/pyslam/thirdparty/monogs/submodules/simple-knn/build/temp.linux-x86_64-cpython-311/simple_knn.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -O2 -allow-unsupported-compiler -std=c++20 -gencode=arch=compute_50,code=sm_50 -gencode=arch=compute_52,code=sm_52 -gencode=arch=compute_53,code=sm_53 -gencode=arch=compute_60,code=sm_60 -gencode=arch=compute_61,code=sm_61 -gencode=arch=compute_62,code=sm_62 -gencode=arch=compute_70,code=sm_70 -gencode=arch=compute_72,code=sm_72 -gencode=arch=compute_75,code=sm_75 -gencode=arch=compute_80,code=sm_80 -gencode=arch=compute_86,code=sm_86 -gencode=arch=compute_87,code=sm_87 -gencode=arch=compute_89,code=sm_89 -gencode=arch=compute_90,code=sm_90 -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=_C -D_GLIBCXX_USE_CXX11_ABI=0
      FAILED: [code=2] /mnt/new/pyslam/thirdparty/monogs/submodules/simple-knn/build/temp.linux-x86_64-cpython-311/simple_knn.o
      /usr/local/cuda-12.2/bin/nvcc --generate-dependencies-with-compile --dependency-output /mnt/new/pyslam/thirdparty/monogs/submodules/simple-knn/build/temp.linux-x86_64-cpython-311/simple_knn.o.d -I/home/meic/anaconda3/envs/pyslam/lib/python3.11/site-packages/torch/include -I/home/meic/anaconda3/envs/pyslam/lib/python3.11/site-packages/torch/include/torch/csrc/api/include -I/home/meic/anaconda3/envs/pyslam/lib/python3.11/site-packages/torch/include/TH -I/home/meic/anaconda3/envs/pyslam/lib/python3.11/site-packages [... cut ...]
        File "/home/meic/anaconda3/envs/pyslam/lib/python3.11/site-packages/torch/utils/cpp_extension.py", line 2112, in _run_ninja_build
          raise RuntimeError(message) from e
      RuntimeError: Error compiling objects for extension
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for simple_knn
  Running setup.py clean for simple_knn
Failed to build simple_knn
error: failed-wheel-build-for-install

× Failed to build installable wheels for some pyproject.toml based projects
╰─> simple_knn
```

luigifreda, 2025-09-01:
Hi,
I was able to reproduce the issue. It seems to be related to building `simple_knn` and `diff-gaussian-rasterization` with C++20 (even if detected as supported). I forced the build to use C++17 (on Ubuntu 22 with CUDA 12.2), and it now works on my side. Let me know how it goes for you.