luigifreda/pyslam #202: AttributeError: module 'pydbow3' has no attribute 'BowVector'

mirpouya, 2025-09-19:
(pyslam) aixlab@aixlab:~/pyslam$ ./main_slam.py 
2025-09-19 16:12:38.125437: I tensorflow/core/util/port.cc:153] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2025-09-19 16:12:38.154353: I tensorflow/core/platform/cpu_feature_guard.cc:210] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX512F AVX512_VNNI AVX512_BF16 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2025-09-19 16:12:38.900734: I tensorflow/core/util/port.cc:153] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
/home/aixlab/anaconda3/envs/pyslam/lib/python3.11/site-packages/f3rm/features/clip/clip.py:6: UserWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html. The pkg_resources package is slated for removal as early as 2025-11-30. Refrain from using this package or pin to Setuptools<81.
  from pkg_resources import packaging
 WARNING: cannot import DelfFeature2D from pyslam.local_features.feature_delf, check the file docs/TROUBLESHOOTING.md
Traceback (most recent call last):
  File "/home/aixlab/pyslam/./main_slam.py", line 39, in <module>
    from pyslam.slam.slam import Slam, SlamState
  File "/home/aixlab/pyslam/pyslam/slam/slam.py", line 39, in <module>
    from .tracking import Tracking
  File "/home/aixlab/pyslam/pyslam/slam/tracking.py", line 53, in <module>
    from pyslam.loop_closing.loop_closing import LoopClosing
  File "/home/aixlab/pyslam/pyslam/loop_closing/loop_closing.py", line 38, in <module>
    from pyslam.loop_closing.loop_detector_configs import LoopDetectorConfigs
  File "/home/aixlab/pyslam/pyslam/loop_closing/loop_detector_configs.py", line 36, in <module>
    from .loop_detector_dbow3 import LoopDetectorDBoW3
  File "/home/aixlab/pyslam/pyslam/loop_closing/loop_detector_dbow3.py", line 70, in <module>
    class LoopDetectorDBoW3(LoopDetectorBase):
  File "/home/aixlab/pyslam/pyslam/loop_closing/loop_detector_dbow3.py", line 110, in LoopDetectorDBoW3
    def db_query(self, global_des: dbow3.BowVector, frame_id, max_num_results=5):
                                   ^^^^^^^^^^^^^^^
AttributeError: module 'pydbow3' has no attribute 'BowVector'
(pyslam) aixlab@aixlab:~/pyslam$ cd scripts/
(pyslam) aixlab@aixlab:~/pyslam/scripts$ cd ..
(pyslam) aixlab@aixlab:~/pyslam$ $ ./scripts/install_thirdparty.sh   
$: command not found
(pyslam) aixlab@aixlab:~/pyslam$ cd scripts/
(pyslam) aixlab@aixlab:~/pyslam/scripts$ ./install_thirdparty.sh 
================================================ 
Building Thirdparty 
================================================ 
================================================ 
Configuring CUDA ... 
HAVE_CUDA=1
CUDA_VERSION: 0
Skipping nvidia toolkit install since CUDA_VERSION is 0 
WITH_PYTHON_INTERP_CHECK: ON 
EXTERNAL_OPTIONS:  -DWITH_PYTHON_INTERP_CHECK=ON -DCMAKE_POLICY_VERSION_MINIMUM=3.5
Currently used pyenv: pyslam 
================================================================== 
Configuring and building thirdparty/orbslam2_features ... 
external option: -DWITH_PYTHON_INTERP_CHECK=ON -DCMAKE_POLICY_VERSION_MINIMUM=3.5
EXTERNAL_OPTIONS: -DWITH_PYTHON_INTERP_CHECK=ON -DCMAKE_POLICY_VERSION_MINIMUM=3.5
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

-- Build type: Release
-- CMAKE_CXX_STANDARD: 17
-- CXX flags:  -O3 -march=native -fPIC -DNDEBUG
-- found OpenCV version: 4.6.0
-- opencv include: /usr/include/opencv4
-- opencv lib dirs: /usr
-- opencv libs: opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_stitching;opencv_video;opencv_videoio;opencv_alphamat;opencv_aruco;opencv_barcode;opencv_bgsegm;opencv_bioinspired;opencv_ccalib;opencv_cvv;opencv_datasets;opencv_dnn_objdetect;opencv_dnn_superres;opencv_dpm;opencv_face;opencv_freetype;opencv_fuzzy;opencv_hdf;opencv_hfs;opencv_img_hash;opencv_intensity_transform;opencv_line_descriptor;opencv_mcc;opencv_optflow;opencv_phase_unwrapping;opencv_plot;opencv_quality;opencv_rapid;opencv_reg;opencv_rgbd;opencv_saliency;opencv_shape;opencv_stereo;opencv_structured_light;opencv_superres;opencv_surface_matching;opencv_text;opencv_tracking;opencv_videostab;opencv_viz;opencv_wechat_qrcode;opencv_ximgproc;opencv_xobjdetect;opencv_xphoto
-- WITH_PYTHON_INTERP_CHECK: ON
CMake Warning (dev) at CMakeLists.txt:30 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PythonInterp: /home/aixlab/anaconda3/envs/pyslam/bin/python (found version "3.11.9")
[... 686 lines cut ...]
2025-09-19 16:14:10.843200: I tensorflow/core/util/port.cc:153] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2025-09-19 16:14:10.872042: I tensorflow/core/platform/cpu_feature_guard.cc:210] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX512F AVX512_VNNI AVX512_BF16 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2025-09-19 16:14:11.635837: I tensorflow/core/util/port.cc:153] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
/home/aixlab/anaconda3/envs/pyslam/lib/python3.11/site-packages/f3rm/features/clip/clip.py:6: UserWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html. The pkg_resources package is slated for removal as early as 2025-11-30. Refrain from using this package or pin to Setuptools<81.
  from pkg_resources import packaging
 WARNING: cannot import DelfFeature2D from pyslam.local_features.feature_delf, check the file docs/TROUBLESHOOTING.md
Traceback (most recent call last):
  File "/home/aixlab/pyslam/./main_slam.py", line 39, in <module>
    from pyslam.slam.slam import Slam, SlamState
  File "/home/aixlab/pyslam/pyslam/slam/slam.py", line 39, in <module>
    from .tracking import Tracking
  File "/home/aixlab/pyslam/pyslam/slam/tracking.py", line 53, in <module>
    from pyslam.loop_closing.loop_closing import LoopClosing
  File "/home/aixlab/pyslam/pyslam/loop_closing/loop_closing.py", line 38, in <module>
    from pyslam.loop_closing.loop_detector_configs import LoopDetectorConfigs
  File "/home/aixlab/pyslam/pyslam/loop_closing/loop_detector_configs.py", line 36, in <module>
    from .loop_detector_dbow3 import LoopDetectorDBoW3
  File "/home/aixlab/pyslam/pyslam/loop_closing/loop_detector_dbow3.py", line 70, in <module>
    class LoopDetectorDBoW3(LoopDetectorBase):
  File "/home/aixlab/pyslam/pyslam/loop_closing/loop_detector_dbow3.py", line 110, in LoopDetectorDBoW3
    def db_query(self, global_des: dbow3.BowVector, frame_id, max_num_results=5):
                                   ^^^^^^^^^^^^^^^
AttributeError: module 'pydbow3' has no attribute 'BowVector'
(pyslam) aixlab@aixlab:~/pyslam$

mirpouya, 2025-09-19:
(pyslam) aixlab@aixlab:~/pyslam$ ./main_slam.py 
2025-09-19 17:43:37.862313: I tensorflow/core/util/port.cc:153] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2025-09-19 17:43:37.890096: I tensorflow/core/platform/cpu_feature_guard.cc:210] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX512F AVX512_VNNI AVX512_BF16 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2025-09-19 17:43:38.637169: I tensorflow/core/util/port.cc:153] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
/home/aixlab/anaconda3/envs/pyslam/lib/python3.11/site-packages/f3rm/features/clip/clip.py:6: UserWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html. The pkg_resources package is slated for removal as early as 2025-11-30. Refrain from using this package or pin to Setuptools<81.
  from pkg_resources import packaging
 WARNING: cannot import DelfFeature2D from pyslam.local_features.feature_delf, check the file docs/TROUBLESHOOTING.md
Traceback (most recent call last):
  File "/home/aixlab/pyslam/./main_slam.py", line 39, in <module>
    from pyslam.slam.slam import Slam, SlamState
  File "/home/aixlab/pyslam/pyslam/slam/slam.py", line 39, in <module>
    from .tracking import Tracking
  File "/home/aixlab/pyslam/pyslam/slam/tracking.py", line 53, in <module>
    from pyslam.loop_closing.loop_closing import LoopClosing
  File "/home/aixlab/pyslam/pyslam/loop_closing/loop_closing.py", line 38, in <module>
    from pyslam.loop_closing.loop_detector_configs import LoopDetectorConfigs
  File "/home/aixlab/pyslam/pyslam/loop_closing/loop_detector_configs.py", line 36, in <module>
    from .loop_detector_dbow3 import LoopDetectorDBoW3
  File "/home/aixlab/pyslam/pyslam/loop_closing/loop_detector_dbow3.py", line 70, in <module>
    class LoopDetectorDBoW3(LoopDetectorBase):
  File "/home/aixlab/pyslam/pyslam/loop_closing/loop_detector_dbow3.py", line 110, in LoopDetectorDBoW3
    def db_query(self, global_des: dbow3.BowVector, frame_id, max_num_results=5):
                                   ^^^^^^^^^^^^^^^
AttributeError: module 'pydbow3' has no attribute 'BowVector'
(pyslam) aixlab@aixlab:~/pyslam$

luigifreda, 2025-09-20:
Hi,
Could you try the following commands and copy&paste the terminal output in a log text file?
```bash
sudo apt update
sudo apt install libcurl4-openssl-dev libgdal-dev libhdf5-dev

cd pyslam  # get into your working copy 
. pyenv-activate 

cd thirdparty/orbslam2_features/
./clean.sh
cd -

cd thirdparty/pydbow3/
./clean.sh
cd - 

cd thirdparty/pyibow/
./clean.sh
cd -

./scripts/install_thirdparty.sh 
```

mirpouya, 2025-09-20:
unfortunately it didn't work:

[log.txt](https://github.com/user-attachments/files/22442806/log.txt)

(pyslam) aixlab@aixlab:~/pyslam$ ./main_slam.py 
2025-09-20 11:10:42.671689: I tensorflow/core/util/port.cc:153] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2025-09-20 11:10:42.699823: I tensorflow/core/platform/cpu_feature_guard.cc:210] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX512F AVX512_VNNI AVX512_BF16 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2025-09-20 11:10:43.451042: I tensorflow/core/util/port.cc:153] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
/home/aixlab/anaconda3/envs/pyslam/lib/python3.11/site-packages/f3rm/features/clip/clip.py:6: UserWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html. The pkg_resources package is slated for removal as early as 2025-11-30. Refrain from using this package or pin to Setuptools<81.
  from pkg_resources import packaging
 WARNING: cannot import DelfFeature2D from pyslam.local_features.feature_delf, check the file docs/TROUBLESHOOTING.md
Traceback (most recent call last):
  File "/home/aixlab/pyslam/./main_slam.py", line 39, in <module>
    from pyslam.slam.slam import Slam, SlamState
  File "/home/aixlab/pyslam/pyslam/slam/slam.py", line 39, in <module>
    from .tracking import Tracking
  File "/home/aixlab/pyslam/pyslam/slam/tracking.py", line 53, in <module>
    from pyslam.loop_closing.loop_closing import LoopClosing
  File "/home/aixlab/pyslam/pyslam/loop_closing/loop_closing.py", line 38, in <module>
    from pyslam.loop_closing.loop_detector_configs import LoopDetectorConfigs
  File "/home/aixlab/pyslam/pyslam/loop_closing/loop_detector_configs.py", line 36, in <module>
    from .loop_detector_dbow3 import LoopDetectorDBoW3
  File "/home/aixlab/pyslam/pyslam/loop_closing/loop_detector_dbow3.py", line 70, in <module>
    class LoopDetectorDBoW3(LoopDetectorBase):
  File "/home/aixlab/pyslam/pyslam/loop_closing/loop_detector_dbow3.py", line 110, in LoopDetectorDBoW3
    def db_query(self, global_des: dbow3.BowVector, frame_id, max_num_results=5):
                                   ^^^^^^^^^^^^^^^
AttributeError: module 'pydbow3' has no attribute 'BowVector'
(pyslam) aixlab@aixlab:~/pyslam$ ./main_slam.py 
2025-09-20 11:49:18.276952: I tensorflow/core/util/port.cc:153] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2025-09-20 11:49:18.305064: I tensorflow/core/platform/cpu_feature_guard.cc:210] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX512F AVX512_VNNI AVX512_BF16 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2025-09-20 11:49:19.055240: I tensorflow/core/util/port.cc:153] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
/home/aixlab/anaconda3/envs/pyslam/lib/python3.11/site-packages/f3rm/features/clip/clip.py:6: UserWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html. The pkg_resources package is slated for removal as early as 2025-11-30. Refrain from using this package or pin to Setuptools<81.
  from pkg_resources import packaging
 WARNING: cannot import DelfFeature2D from pyslam.local_features.feature_delf, check the file docs/TROUBLESHOOTING.md
Traceback (most recent call last):
  File "/home/aixlab/pyslam/./main_slam.py", line 39, in <module>
    from pyslam.slam.slam import Slam, SlamState
  File "/home/aixlab/pyslam/pyslam/slam/slam.py", line 39, in <module>
    from .tracking import Tracking
  File "/home/aixlab/pyslam/pyslam/slam/tracking.py", line 53, in <module>
    from pyslam.loop_closing.loop_closing import LoopClosing
  File "/home/aixlab/pyslam/pyslam/loop_closing/loop_closing.py", line 38, in <module>
    from pyslam.loop_closing.loop_detector_configs import LoopDetectorConfigs
  File "/home/aixlab/pyslam/pyslam/loop_closing/loop_detector_configs.py", line 36, in <module>
    from .loop_detector_dbow3 import LoopDetectorDBoW3
  File "/home/aixlab/pyslam/pyslam/loop_closing/loop_detector_dbow3.py", line 70, in <module>
    class LoopDetectorDBoW3(LoopDetectorBase):
  File "/home/aixlab/pyslam/pyslam/loop_closing/loop_detector_dbow3.py", line 110, in LoopDetectorDBoW3
    def db_query(self, global_des: dbow3.BowVector, frame_id, max_num_results=5):
                                   ^^^^^^^^^^^^^^^
AttributeError: module 'pydbow3' has no attribute 'BowVector'
(pyslam) aixlab@aixlab:~/pyslam$

luigifreda, 2025-09-21:
Merged that branch into master.